chrisb2/pyb_ina219 #1: LoPy ESP32 Port

JamesMcCarthy79, 2018-05-16:
Hi,

I'm trying to use your library for a pycom LoPy and I'm getting errors on "line 12" specifically to do with the import logging. Should I be copying across the logging.mpy file also or is there another way of calling the logging with pycom?

chrisb2, 2018-05-16:
Yes, I think that is right. Looking at the readme it does mention copying logging, but only in the [Debugging](https://github.com/chrisb2/pyb_ina219#debugging) section, I may need to update it, will check tomorrow.

Let me know how you get on.

regards,
Chris

chrisb2, 2018-05-16:
Looks like this is may be the relevant dos on setup I2C:
[I2C](https://docs.pycom.io/chapter/tutorials/all/i2c.html)

JamesMcCarthy79, 2018-05-16:
Yeah, I'm all good with I2C implementation just not sure how to implement the logging call. 
I did try to add the logging.mpy but not sure if its formatted right. Is the .mpy file for esp8266 chipset or can it be used for esp32 also? I'm guessing it can as you have used library with Lolin32 Lite.

JamesMcCarthy79, 2018-05-17:
Ok, so logging library all good I think there is some incompatibility with the argument structure I'm not exactly sure of correct arguments yet but once corrected will post.

[image]

chrisb2, 2018-05-17:
In ina219 constructor you have case of i2c argument wrong, should be lower
case.

chrisb2, 2018-05-17:
Should not be a problem, but baudrate is really low, maybe you missed off a zero? Default baudrate for lopy is [100000](https://docs.pycom.io/chapter/firmwareapi/pycom/machine/I2C.html#i2cinitmode---baudrate100000-pinssda-scl)

JamesMcCarthy79, 2018-05-17:
case doesn't make a difference as does the baudrate still get same error.

chrisb2, 2018-05-17:
The failing line is [391](https://github.com/chrisb2/pyb_ina219/blob/1340e38b7e7959a6b2a0467d12f018f48850e341/ina219.py#L391) and comparing docs for [pyboard](http://docs.micropython.org/en/latest/pyboard/library/machine.I2C.html#machine.I2C.writeto_mem) and [Pycom](https://docs.pycom.io/chapter/firmwareapi/pycom/machine/I2C.html#i2cwritetomemaddr-memaddr-buf--addrsize8), the signature is the same except pycom returns number of bytes written, but code does not use return value. Might have to compare source code for the method between ports.

chrisb2, 2018-05-17:
Lets try something simple, reading the default config of ina219. In REPL set up i2c variable as per your code then:
```
i2c.readfrom_mem(0x40, 0x00, 2)
```

JamesMcCarthy79, 2018-05-17:
Yep I get b'9\x9f' response

chrisb2, 2018-05-17:
OK how about writing with:
```
ina = INA219(0.1, i2c)
ic2.writeto_mem(0x40, 0x00, ina.__to_bytes(0x8333))
```

chrisb2, 2018-05-17:
The default config value b'9\x9f' (decimal 14751) is the expected power on default value for the ina219 (see p25 of data sheet).

JamesMcCarthy79, 2018-05-17:
what should I be seeing here?

chrisb2, 2018-05-17:
Doc says the return value is the number of bytes written, so if it worked I would expect it to return 2. Also if you do this then read the config again you should get back the same value. I used this to convert byte array to int:
```
int.from_bytes(b'9\x9f', 'big')
```
So if you re-read config you should get 33587 (0x8333)

JamesMcCarthy79, 2018-05-17:
Here is my output

[image]

JamesMcCarthy79, 2018-05-17:
Ok it was the byte number, increased to 10

b'9\x9f\xff\xff\xff\xff\xff\xff\xff\xff'

chrisb2, 2018-05-17:
Sorry, not clear what that it the result of?

JamesMcCarthy79, 2018-05-17:
`i2c = I2C(0, I2C.MASTER, baudrate=100000)
ina = INA219(0.1, i2c)
tmp = i2c.readfrom_mem(0x40, 0x00, 2)
print(tmp)
gc.enable()
gc.collect()
i2c.writeto_mem(0x40, 0x00, ina.__to_bytes(0x8333))
print(i2c.writeto_mem(0x40, 0x00, ina.__to_bytes(0x8333)))
tmp2 = i2c.readfrom_mem(0x40, 0x00, 20)
print(tmp2)`

chrisb2, 2018-05-17:
You have a typo:
```
tmp2 = i2c.readfrom_mem(0x40, 0x00, 20) 
```
last arg should be 2.

JamesMcCarthy79, 2018-05-17:
```
i2c = I2C(0, I2C.MASTER, baudrate=100000)
ina = INA219(0.1, i2c)
tmp = i2c.readfrom_mem(0x40, 0x00, 2)
print(tmp)
i2c.writeto_mem(0x40, 0x00, ina.__to_bytes(0x8333))
tmp2 = i2c.readfrom_mem(0x40, 0x00, 2)
print(tmp2)
```
I get this 
b'9\x9f'
b'9\x9f'

chrisb2, 2018-05-17:
Can you put a 1ms sleep before:
```
print(i2c.writeto_mem(0x40, 0x00, ina.__to_bytes(0x8333)))
```
Spec talks about reading too quickly after writing, but only at 1MHz, but lets see what happens anyway.

JamesMcCarthy79, 2018-05-17:
```
i2c = I2C(0, I2C.MASTER, baudrate=100000)
ina = INA219(0.1, i2c)
tmp = i2c.readfrom_mem(0x40, 0x00, 2)
print("Read 1 :" +str(tmp))
i2c.writeto_mem(0x40, 0x00, ina.__to_bytes(0x8333))
time.sleep(1)
tmp2 = i2c.readfrom_mem(0x40, 0x00, 2)
print("Read 2 :" +str(tmp2))
```
Response
Read 1 :b'9\x9f'
Read 2 :b'9\x9f'

chrisb2, 2018-05-17:
OK maybe there is something fishy with ina.__to_bytes(), lets get rid of it:
```
i2c = I2C(0, I2C.MASTER, baudrate=100000)
tmp = i2c.readfrom_mem(0x40, 0x00, 2)
print("Read 1 :" +str(tmp))
num = i2c.writeto_mem(0x40, 0x00, b'\x833')
print(num)
time.sleep(1)
tmp2 = i2c.readfrom_mem(0x40, 0x00, 2)
print("Read 2 :" +str(tmp2))
```

JamesMcCarthy79, 2018-05-17:
Response
Read 1 :b'9\x9f'
2
Read 2 :b'9\x9f'

chrisb2, 2018-05-17:
This is getting mysterious, either  b'\x833' is not a valid config value and is therefore not applied, or write operations are simply not working.

I am going to connect up my ina219 breakout to an esp32 and check a few things out, will get back to you probably over the weekend.

JamesMcCarthy79, 2018-05-17:
Thanks Chris appreciate the help.

chrisb2, 2018-05-18:
I understand the reason we could not update configuration register; 0x8333 is simply not a valid value and is ignored on my esp32 with std Micropython 1.9.4.

From the error msg we know that one of the 3 args to writeto_mem() must be the wrong type, so I have put together a series of calls which try various values for them. Note that these are writing to the calibration register (the one that is failing in your script). Give them a try and let me know which succeed and which fail, hopefully this will tell us exactly which arg is the problem.
```
#setup
i2c = I2C(0, I2C.MASTER, baudrate=100000)
ina = INA219(0.1, i2c)

i2c.writeto_mem(0x40, 0x05, bytearray([0x83, 0x33]))
i2c.readfrom_mem(0x40, 0x05, 2)
# expected: b'\x832'

i2c.writeto_mem(0x40, 0x05, bytes([0x83, 0x00]))
i2c.readfrom_mem(0x40, 0x05, 2)
# expected: b'\x83\x00'

i2c.writeto_mem(0x40, 0x05, ina.__to_bytes(0x8333))
i2c.readfrom_mem(0x40, 0x05, 2)
# expected: b'\x832'

i2c.writeto_mem(ina._address, 0x05, ina.__to_bytes(0x8300))
i2c.readfrom_mem(0x40, 0x05, 2)
# expected: b'\x83\x00'

i2c.writeto_mem(ina._address, ina.__REG_CALIBRATION, ina.__to_bytes(0x8333))
i2c.readfrom_mem(0x40, 0x05, 2)
# expected: b'\x832'

ina.__write_register(ina.__REG_CALIBRATION, 0x8300)
i2c.readfrom_mem(0x40, 0x05, 2)
# expected: b'\x83\x00'

ina._calibration_register(0x8333)
i2c.readfrom_mem(0x40, 0x05, 2)
# expected: b'\x832'

ina._calibrate(16, 0.04, 0.4)
i2c.readfrom_mem(0x40, 0x05, 2)
# expected: b'\x832'
```

JamesMcCarthy79, 2018-05-18:
I left in a hurry today and left my bag at the office which has my dev boards in it. I hopefully have one lying around the house I can cobble together otherwise I wont be able to test until Monday.

JamesMcCarthy79, 2018-05-21:
Hi Chris,

See results below

```
#setup
i2c = I2C(0, I2C.MASTER, baudrate=100000)
ina = INA219(0.1, i2c)

i2c.writeto_mem(0x40, 0x05, bytearray([0x83, 0x33]))
i2c.readfrom_mem(0x40, 0x05, 2)
# expected: b'\x832'
## Response: b'\x832'

i2c.writeto_mem(0x40, 0x05, bytes([0x83, 0x00]))
i2c.readfrom_mem(0x40, 0x05, 2)
# expected: b'\x83\x00'
## Response: b'\x83\x00'

i2c.writeto_mem(0x40, 0x05, ina.__to_bytes(0x8333))
i2c.readfrom_mem(0x40, 0x05, 2)
# expected: b'\x832'
## Response: b'\x832'

i2c.writeto_mem(ina._address, 0x05, ina.__to_bytes(0x8300))
i2c.readfrom_mem(0x40, 0x05, 2)
# expected: b'\x83\x00'
## Response: b'\x83\x00'

i2c.writeto_mem(ina._address, ina.__REG_CALIBRATION, ina.__to_bytes(0x8333))
i2c.readfrom_mem(0x40, 0x05, 2)
# expected: b'\x832'
## Response: b'\x832'

ina.__write_register(ina.__REG_CALIBRATION, 0x8300)
i2c.readfrom_mem(0x40, 0x05, 2)
# expected: b'\x83\x00'
## Response: b'\x83\x00'

ina._calibration_register(0x8333)
i2c.readfrom_mem(0x40, 0x05, 2)
# expected: b'\x832'
## Response: b'\x832'

ina._calibrate(16, 0.04, 0.4)
i2c.readfrom_mem(0x40, 0x05, 2)
# expected: b'\x832'
## Response: b'\x832'
```

JamesMcCarthy79, 2018-05-21:
It works fine without anything connected to it.
[image]
however as soon as i connect anything to the INA219 it exceeds the current device limit which doesn't make sense. Is it to do with the SHUNT_OHMS setting?

[image]

chrisb2, 2018-05-21:
OK this is looking better, the previous type error has gone, not sure why, did you change anything?

What you are getting now is expected behavior if too much current is flowing, from the logging that is more than 3.2A. How much current are you expecting to be flowing? 

The ina219 is actually not measuring current directly, but measuring the voltage drop across the resistor and it can measure a maximum voltage drop (gain) of 0.32V, so via I = V/R = 0.32/0.1 thats 3.2A.

To enable the ina219 to measure larger currents you will need to replace the 0.1 Ohm resistor with a smaller value, for example if the value was 0.01 Ohm you would be able to measure 32A. With this change the resolution would drop from 0.8mA to 8mA. A suitable resistor would be something like a [0.01 Ohm, 1W](https://au.rs-online.com/web/p/surface-mount-fixed-resistors/0566197/).

chrisb2, 2018-05-21:
By the way the setup of the device in lines 29-31 would be better outside your while loop, setting up once is fine.

JamesMcCarthy79, 2018-05-21:
The aim is to read from a 2 wire flow sensor which outputs 24v loop at 4ma-20ma so this will be fine I just don't have access to any 2 wire sensors atm so I am going to build a circuit to simulate this tomorrow morning and will let you know.

JamesMcCarthy79, 2018-05-21:
Ignore the while loop I just had that whilst I was playing around with a dc voltage regulator but it wouldn't do less than 3.5A so I need to build the test cct above. 

[image]

chrisb2, 2018-05-24:
Cool, that is great to hear. I would like to put a simple version of your program on the [wiki](https://github.com/chrisb2/pyb_ina219/wiki) as an example of using this library with LoPy, is that OK with you?

Chris

JamesMcCarthy79, 2018-05-24:
Yes of course hopefully can help out anyone else looking to use your library

closed 2019-05-11 by chrisb2