chrisb2/pyb_ina219 #9: (Solution) "OSError: [Errno 5] EIO" when running with Micropython on Raspberry Pi Pico

mburde7, 2022-08-17:
Hi Chris, 

I tried to get this running on my Raspberry Pi Pico and was getting an error. Here is the message: 

"Traceback (most recent call last):
  File "<stdin>", line 8, in <module>
  File "ina219.py", line 189, in configure
  File "ina219.py", line 316, in _calibrate
  File "ina219.py", line 361, in _calibration_register
  File "ina219.py", line 391, in __write_register
OSError: [Errno 5] EIO"

From this thread: https://forum.micropython.org/viewtopic.php?t=4746 I saw that the Pico seems to need delays for I2C to work well, so I added the utime.sleep delays to your example code:

~~~~~~~
from ina219 import INA219
from machine import I2C
import utime

I2C_INTERFACE_NO = 1
SHUNT_OHMS = 0.1  # Check value of shunt used with your INA219

ina = INA219(SHUNT_OHMS, I2C(I2C_INTERFACE_NO, scl=machine.Pin(19), sda=machine.Pin(18))) 
utime.sleep_ms(10) 
ina.configure()
utime.sleep_ms(10)
print("Bus Voltage: %.3f V" % ina.voltage())
print("Current: %.3f mA" % ina.current())
print("Power: %.3f mW" % ina.power())
~~~~~~

Hope this helps anyone else with similar issues!

IHasToes1234, 2022-09-06:
I'm having this same issue and am wondering if you ever figured it out?

mburde7, 2022-09-06:
Maybe my comment wasn't super clear - the "utime.sleep_ms(10)" code additions do fix the issue and I haven't had any trouble with the INA219 since.

IHasToes1234, 2022-09-07:
I got this error when trying the code from your comment:

Traceback (most recent call last):
  File "<stdin>", line 10, in <module>
  File "ina219.py", line 189, in configure
  File "ina219.py", line 316, in _calibrate
  File "ina219.py", line 361, in _calibration_register
  File "ina219.py", line 391, in __write_register
OSError: [Errno 5] EIO

Do you know what could be causing this?

mburde7, 2022-09-10:
Can you give more details on your setup? Maybe your circuit design. Also make sure you're using the right pins - I'm using pins GP 18 and GP 19 which are actually the 24th and 25th pico pins respectively.

IHasToes1234, 2022-09-10:
Worked perfectly when I used GP18 & 19. If I were looking to connect another INA219 which pins would you recommend?

mburde7, 2022-09-11:
Glad its working! There are two I2C bus connections you can use (this is using I2C 1). In my setup I'm using two INA219s - one connected to GP18, GP19 and the other connected to GP12, GP13. Just make sure you indicate which I2C bus you're using by adding the 0 or 1 to this line (reference the pico pinout to see where the I2C0 and I2C1 are). Here is what I am using:

ina = INA219(SHUNT_OHMS, I2C(1, scl=machine.Pin(19), sda=machine.Pin(18)))
ina_1 = INA219(SHUNT_OHMS, I2C(0, scl=machine.Pin(13), sda=machine.Pin(12)))

IHasToes1234, 2022-10-06:
Hey, I know its been a while, but I need help again. Do I need a second copy of every file for the second INA board? I would assume yes, and if so, what do I need to change in the second set of files? I changed the address in the ina219.py file and in my run file. I still cant seem to get it to work.

Here is my run code if needed (starting on button press, checking every hour):
`from machine import Pin
import _thread
import utime

button=Pin(0, Pin.IN, Pin.PULL_DOWN)

def tp():
    execfile("run.py")  
def sp():
    execfile("run_1.py")
time=5
buttonNow=0
while True:
    if buttonNow==1:
        time=time+1
        print("Tracking Panel at %s:30:" % time)
        tp()
        print("Stationary Panel at %s:30:" % time)
        sp()
        utime.sleep(2.5)
        buttonNow=1
    else:
        buttonNow=button.value()`

Run_1 file (if needed):
`from ina219_1 import INA219
from ina219_1 import DeviceRangeError
from machine import I2C
import utime

#data list (clear daily and add to sheet)
data = []

I2C_INTERFACE_NO = 0
SHUNT_OHMS = 0.1  # Check value of shunt used with your INA219

ina = INA219(SHUNT_OHMS, I2C(I2C_INTERFACE_NO, scl=machine.Pin(13), sda=machine.Pin(12)), address=0x41) 
utime.sleep_ms(10) 
ina.configure()
utime.sleep_ms(10)
try:
    data = ['Bus Voltage: %.3f V' % ina.voltage(), 'Bus Current: %.3f mA' % ina.current(),'Shunt voltage: %.3f mV' % ina.shunt_voltage(),'Power: %.3f mW' % ina.power()]
    print(data)
except DeviceRangeError as e:
    # Current out of device range with specified shunt resister
    print ("e")`

Error:
"Traceback (most recent call last):
  File "<stdin>", line 19, in <module>
  File "<stdin>", line 10, in sp
  File "run_1.py", line 14, in <module>
  File "ina219_1.py", line 186, in configure
  File "ina219_1.py", line 313, in _calibrate
  File "ina219_1.py", line 358, in _calibration_register
  File "ina219_1.py", line 388, in __write_register
OSError: [Errno 5] EIO"

PS: Thanks for all the help!

IHasToes1234, 2022-10-08:
Also if you dont feel like reading this I could just look at your code and figure it out.

chrisb2, 2022-10-12:
So you are talking to two ina219's from one pico?

In this case one of them need a different address on i2c bus, see section
8.5.5.1 of the datasheet. Then in code each ina219 instance can be
initialize with its own address.

Regards
Chris

IHasToes1234, 2022-10-12:
Hey Chris! 

I am very dumb and have no idea what you mean. I soldered the pins on the breakout board to make the address 0x41 but have no idea what to do code wise. Do I need a separate ina219.py, and if so, do I only have to change the address for the second. I believe they are on different i2c buses as well.

chrisb2, 2022-10-13:
Assuming you have correctly configured the two devices with different addresses, you can attach them both to the same i2c bus (same two i2c sda/scl pins). You only need one ina219.py file and import.

The address can be passed in as an argument to the constructor, see *address* arg: [https://github.com/chrisb2/pyb_ina219#functions](https://github.com/chrisb2/pyb_ina219#functions) 

I suggest you separately connect each device and check its responding correctly on its address, then attach both and test.

Chris

supreeet, 2022-11-06:
I'm getting errors when a relay connects and disconnects the sensor to a battery and the code runs in a loop. Nothing is connected on the other side but it says current overflow. I've tried adding delays but still randomly after sometime I get these errors. 

Traceback (most recent call last):
  File "<stdin>", line 47, in <module>
  File "ina219.py", line 219, in power
  File "ina219.py", line 381, in _power_register
  File "ina219.py", line 397, in __read_register
OSError: [Errno 19] ENODEV
 
or..

Traceback (most recent call last):
  File "<stdin>", line 48, in <module>
  File "ina219.py", line 226, in shunt_voltage
  File "ina219.py", line 255, in _handle_current_overflow
  File "ina219.py", line 364, in _has_current_overflow
  File "ina219.py", line 372, in _read_voltage_register
  File "ina219.py", line 397, in __read_register
OSError: [Errno 5] EIO

Also what does "The library currently only supports continuous reads of voltage and power, but not triggered reads" means, I can't use it like this?

chrisb2, 2023-03-28:
No comments for a while, so closing, please raise a new issue if required. Thx

closed 2023-03-28 by chrisb2