chrisb2/pyb_ina219 #10: I'm getting errors when a relay connects and disconnects the sensor to a battery and the code runs in a loop. Nothing is connected on the other side but it says current overflow. I've tried adding delays but still randomly after sometime I get these errors.

supreeet, 2022-11-09:
I'm getting errors when a relay connects and disconnects the sensor to a battery and the code runs in a loop. Nothing is connected on the other side but it says current overflow. I've tried adding delays but still randomly after sometime I get these errors. 

Its a Pico W

Traceback (most recent call last):
  File "<stdin>", line 47, in <module>
  File "ina219.py", line 219, in power
  File "ina219.py", line 381, in _power_register
  File "ina219.py", line 397, in __read_register
OSError: [Errno 19] ENODEV
 
or..

Traceback (most recent call last):
  File "<stdin>", line 48, in <module>
  File "ina219.py", line 226, in shunt_voltage
  File "ina219.py", line 255, in _handle_current_overflow
  File "ina219.py", line 364, in _has_current_overflow
  File "ina219.py", line 372, in _read_voltage_register
  File "ina219.py", line 397, in __read_register
OSError: [Errno 5] EIO

Also what does "The library currently only supports continuous reads of voltage and power, but not triggered reads" means, I can't use it like this?

_Originally posted by @supreeet in https://github.com/chrisb2/pyb_ina219/issues/9#issuecomment-1304801079_

chrisb2, 2022-11-17:
Sorry for delay in reply. Please can you sketch your circuit and attach, it's possible electrical noise from relay operation is causing an issue.

Triggered read refers to functionality which allows single shot as opposed to continuous reads, you can read about it in the ina219 datasheet. 

Chris

supreeet, 2022-11-18:
Okay thanks and here is the simple circuit. I'm using the common 3.2a ina219 breakout board with pull resistors and few tiny caps already there, the battery is 12.8v nominal lifepo4
[image]

chrisb2, 2022-11-21:
I did a bit of research on relay noise and came up with [Snubber Circuits for Protecting Relay Contacts from Arcing](https://makingcircuits.com/blog/snubber-circuits-for-protecting-relay-contacts-from-arcing/). These ideas are easy and harmless to try. I am not an electronics engineer so if this does not work I suggest you ask on [electronics.stackexchange.com](https://electronics.stackexchange.com/).

Chris

supreeet, 2022-11-21:
Okay, thanks a lot!

closed 2022-11-22 by supreeet