chrisb2/pyb_ina219 #11: Measurements not correct from INA219 on a RPi Pico

mebj, 2023-03-26:
During test of a custom made data logger I have found that values generated by the Pico is more than 6 times lower than when measured by a multimeter and also from an Arduino Nano using the Adafruit library.
Same INA219 and simple test circuit is used for both the Pico and the Nano tests.

I am waiting for another Pico but the unit used for testing has worked well so far so I doubt the problem is there.

A text file with some more info is attached.
[Ina219.txt](https://github.com/chrisb2/pyb_ina219/files/11072522/Ina219.txt)

chrisb2, 2023-03-27:
So just to confirm despite the 'errors' in vscode the code does execute?

As a first step can you attach a sketch of your circuit? One thing to confirm is that you have a common ground between the load under test and the mcu ground.

Do you have specific plug-ins installed in vscode?

Regards,
Chris

mebj, 2023-03-27:
Hi Chris,

Many thanks for a quick answer.

The code does not execute unless the second error is fixed. I made a wild guess and changed "if logging._level == logging.DEBUG:" to "if logging._level_dict == logging.DEBUG:"  and then the code executes.

The entire setup is quite complex because a 20x4 lcd, a DS3231 clock and a sd-card reader/writer module are attached. The INA219 part is simple though because it is the only component attached to I2C1 SDA and I2C1 SDB at GP2 and GP3, other I2C components are on I2C0 at GP16 and GP17 and the sd- card is attached to SPI1 at GP10, 11, 12 and 13.

The INA219 has a common ground with the tested circuit. Without that the measurements were all over the scale but are now well aligned as you can see from the recorded values.

I use Pico-W-Go and Pylance extensions for vscode (and Python of course)

There is another INA219 library (https://github.com/robert-hh/INA219) used by this example (https://scruss.com/blog/2022/08/08/ina219-current-sensor-and-micropython/) and when running this exactly the same erroneous result is obtained.

So there seems to be something very specific with the Pico. I will test another unit once I get hold of one but do not expect any change since the current one is working for other things I have tested.

Best regards.

Mats

chrisb2, 2023-03-28:
Thanks for all the info, I have a pico and vscode, so can look into this, but I am not going to get much time until 7th April.

mebj, 2023-03-28:
That is OK, will be waiting. As I do not know the code or vscode debugging it is probably beyond my capabilities to trace the calculation of the shunt voltage but I can take a look at it while waiting.

chrisb2, 2023-04-01:
Good to hear, will investigate code issue next weekend. 

Chris

chrisb2, 2023-04-11:
Coincidently this [PR](https://github.com/chrisb2/pyb_ina219/pull/12) should fix the issue on L409. 

In vscode I am not currently seeing the error at L258. I have vscode v1.77.1, Pico-W-Go v3.0.1, Pylance v2023.4.10 and Python v2023.6.0.

closed 2023-08-03 by chrisb2