chrisb2/pyb_ina219 #13: ESP8266 issue

MrGadzet, 2023-08-01:
Hi, 
i am trying to run this current sensor with esp8266, but i've got a lot of issues.  I solved a most of the problems myself but one i can't do for 3 days.
During start program it shows me "AttributeError: 'module' object has no attribute '_level'". Code is from your example. You can see it below. The logging.py file from official micropython github from linked file in README.
[image]

`from machine import Pin, I2C
from ina219 import INA219
from logging import INFO

i2c = I2C(scl=Pin(5), sda=Pin(4), freq=400000)

SHUNT_OHMS = 0.1

ina = INA219(SHUNT_OHMS, i2c)
ina.configure()
print("Bus Voltage: %.3f V" % ina.voltage())
print("Current: %.3f mA" % ina.current())
print("Power: %.3f mW" % ina.power())`

Have you got any solution of this problem?

chrisb2, 2023-08-02:
I have not seen this before, something wrong with logging module would be my suspicion, what happens if at the REPL prompt you do:
```
import logging
```
Does this work?

If it does what does:
```
dir(logging)
```
Give you?

Chris

chrisb2, 2023-08-02:
Investigating a bit more I can see that the logging module has been updated recently and the `_level` attribute remove. In the commit history [https://github.com/micropython/micropython-lib/commits/master/python-stdlib/logging/logging.py](https://github.com/micropython/micropython-lib/commits/master/python-stdlib/logging/logging.py), you can see this in the 1st commit on 21st Dec 2022, [https://github.com/micropython/micropython-lib/commit/58bab0add3c730842d39cdfcd6ff5a73ebba15d3](https://github.com/micropython/micropython-lib/commit/58bab0add3c730842d39cdfcd6ff5a73ebba15d3).

So until I can update pyb_in219 to account for this, you can try versions of the logging library before this commit, perhaps the one of 27th Mar 2021 would be a good place to start, so download and install this [https://raw.githubusercontent.com/micropython/micropython-lib/fa13cbbc8b53e2ec6abbb1b0fa33d085582f3c54/python-stdlib/logging/logging.py](https://raw.githubusercontent.com/micropython/micropython-lib/fa13cbbc8b53e2ec6abbb1b0fa33d085582f3c54/python-stdlib/logging/logging.py)

Chris

chrisb2, 2023-08-02:
Update to pyb_ina219 to support latest version of `logging` looks like it might be fairly simple, will try at the weekend.

chrisb2, 2023-08-03:
Fix for this already in progress on [PR12](https://github.com/chrisb2/pyb_ina219/pull/12), am looking to complete this.

MrGadzet, 2023-08-06:
Thanks for quick reply. I tried with the commit you sent and it works.

aiot-maker, 2023-08-07:
I faced same issue reported in the OP when testing the library on a Pico W.  Applied suggestion from below link (chnge in line 409) to got it working.

https://github.com/chrisb2/pyb_ina219/pull/12/commits/8caa4e2243767ff250dacf7982588d915e6f8a88